jneilliii/OctoPrint-RTMPStreamer #2: cant get the plugin to work but docker works

Bretthansenpc, 2018-08-16:
hello I love this plugin but i cant seem to get it to work with the start stop button it says its going but doesnt stream, when I click stop it shows an error saying theres no stream named that. 

although if I run the docker code it runs fine and streams any ideas?

jneilliii, 2018-08-16:
What do you have as your webcam stream setting?  Because docker runs in it's own instance it has to be a hardcoded url so if you use something like the following it should work.  Once you change the webcam stream setting make sure to restart OctoPrint just to be safe.

```
http://192.168.0.2/webcam/?action=stream
```

Bretthansenpc, 2018-08-16:
i have it hard coded in the rtmp tab does it need to be hard coded in the settings for the webcam itself?

Bretthansenpc, 2018-08-16:
got it thanks!!!

closed 2018-08-16 by Bretthansenpc