jneilliii/OctoPrint-RTMPStreamer #4: not working

MikeUSNRet, 2018-08-18:
not working using a usb cam.. does this only work with a pi cam?

jneilliii, 2018-08-19:
I tested with both.  It should work with any camera that works with the default octoprint control tab.  Did you follow the instructions under [Prequisites for Streaming](https://github.com/jneilliii/OctoPrint-RTMPStreamer#prerequisites-for-streaming) section? The most commonly missed setting is setting the webcam stream url in OctoPrint's settings to include a fully qualified url that includes the ip address of your pi.

MikeUSNRet, 2018-08-19:
Using a Raspberry Pi 3
Logitech c270 USB Webcam

I have:
Installed Docker
Cloned and built docker image
Have run this Test command line

*** note.  zzz is my rasp - pi ip addy and the xxxx-xxxx is my youtube stream key ***

docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.zzz/webcam/?action=stream 640x480 30 rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx null 

and this is what I get reported back:

docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.zzz/webcam/?action=stream 640x480 30 rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx null
OctoPrint Webcam Stream: http://192.168.1.zzz/webcam/?action=stream
Stream Resolution: 640x480
Framerate 30
Stream URL posting to: rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx
ffmpeg version N-90990-g652b857b94 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 18.100 / 56. 18.100
  libavcodec     58. 19.101 / 58. 19.101
  libavformat    58. 13.101 / 58. 13.101
  libavdevice    58.  4.100 / 58.  4.100
  libavfilter     7. 21.100 /  7. 21.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
[mjpeg @ 0x235efa0] unable to decode APP fields: Invalid data found when processing input
Input #0, mjpeg, from 'http://192.168.1.xxx/webcam/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 640x480, 30 tbr, 1200k tbn, 30 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx: Input/output error

[mjpeg @ 0x283bfa0] unable to decode APP fields: Invalid data found when processing input 

In Octoprint, this is how I have my RTMP Stream URL entered:
rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx

am able to view webcam in octoprint and record timelapse video so I know that is working well.+

jneilliii, 2018-08-19:
What's the url you have in octoprint's webcam settings for stream? From that error it looks like it's having issues reaching the stream.

MikeUSNRet, 2018-08-19:
in Octoprint settings => Features => Webcam & Timelapse I have this:    /webcam/?action=stream    When I click on Test I correctly see the image that it should be.

Now, I have attempted to use the plugin and click on Start to begin a stream, but after a bit, I get a dialogue box that displays:

RTMP Streamer Error
404 Client Error: Not Found ("No such container: 46bfefcf07d599541a44488c80de1a96d087df2

As mentioned in last post, I ran thru the Prereq for streaming section and can't even get the Test command line this command line to work without getting the info I posted for you:

 docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.zzz/webcam/?action=stream 640x480 30 rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx null

Looks like just one of those things I'll have to chalk up as non-operational with my setup.

jneilliii, 2018-08-19:
Yeah, we need to get the test command working before we'll be able to get the plugin to work.  Try the test command with a local framerate and see if that makes a difference, something like this.  One thing of note, you'll have to change the Octoprint settings => Features => Webcam & Timelapse stream url to be the full ip url `http://192.168.1.zzz/webcam/?action=stream` for the plugin to work.
```
docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.zzz/webcam/?action=stream 640x480 5 rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx null
```
BTW, I did write the YouTube-Live plugin for OctoPrint and this was just a fork of that plugin to work with any generic rtmp service.

MikeUSNRet, 2018-08-19:
putting my info in where zzz and xxxxx's reside and running the script line from within putty, here is the resulting reported info:

 Keep getting this looping now BUT was able to see a live (very much time delayed image) on youtube:

 Last message repeated 2 times
frame=  744 fps=5.0 q=-0.0 size=   12378kB time=00:02:28.44 bitrate= 683.1kbits/[mjpeg @ 0x2b98880] unable to decode APP fields: Invalid data found when processing input

Framerate I was using before was 30,  probably too high for a 640x480 resolution?

MikeUSNRet, 2018-08-19:
re-ran with framerate of 30 and worked fine..now that I know the test line runs, time to see if a live run from within octoprint will run :)

jneilliii, 2018-08-19:
Yeah, it's all dependent on your connection speed and device doing the work (pi 3 vs pi zero,  etc.). With the youtube live plugin, which this one is based on. I force that resolution and frame rate and it's not configurable. I see about a 30-45 second delay as it streams as you mention. Not sure if there's much I can do to make that work any better as it's not a direct push to youtube, but it's reading the mjpg stream and converting it on the fly.

jneilliii, 2018-08-19:
Just saw your last post, good news. Make sure you updated the Octoprint settings => Features => Webcam & Timelapse stream url to match the one in your test command. Also you'll want to make sure to remove the docker image tou ran during test. All in the instructions.

closed 2018-08-19 by jneilliii

MikeUSNRet, 2018-08-19:
hmmm...well, from within putty, the test line connects and streams just fine..  now within octoprint, I click on Start from within the RTMP Plugin and begin printing and nothing happens at all.  it never does start   just a spinning circle thingy in the Start button of the plugin and no live feed ever comes up on the youtube channel.
 Under Webcam & Timelapse I have this:  http://192.168.1.zzz/webcam/?action=stream  I click on Test and see what I"m supposed to so that is good:

under Plugins => RTMP Streamer I have this:

View URL:  https://www.youtube.com/channel/sjroeruJORUOER (whatever my channel is)
RTMP Stream URL:   rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx
Stream Resolution:  640x480
Stream Framerate: 30  (have tried 5 - 30 in increments of 5)
Stream While Printing  (checked)

MikeUSNRet, 2018-08-19:
and my print just finished...and even though nothing ever started streaming, this pops up:

RTMP Streamer Error
404 Client Error: Not Found ("No such container: 46bfefcf07d599541a44488c80de1a96d087df2

jneilliii, 2018-08-19:
That last error is because the print finished and it's trying to close the docker instance that never started because you have stream while printing enabled. I'd recommend restarting octoprint and try pressing the start button on the tab to see if that streams or not.