jneilliii/OctoPrint-RTMPStreamer #5: Not Streaming

BeerGeekGamer, 2018-08-28:
Hello,

I followed your prerequisite and setup instructions, but the stream doesn't seem to be working for me. When I hit the stop button, I do receive the following error.

`RTMP Streamer Error
404 Client Error: Not Found ("No such container: 3cd457ce14753a5aa1080baadc8ea4dc8e0a64ca2a30cda4eb6ef06725e0b0fc")`

When I check the octoprint.log I do see the following entry.

`octoprint.plugins.rtmpstreamer - ERROR - 404 Client Error: Not Found ("No such container: RTMPStreamer")`

I am not familiar with docker, do I need to run a command before hitting the start button?

jneilliii, 2018-08-28:
No, the start button will start the docker image for you. The most common issue is setting the webcam stream url in octoprint's webcam and timelapse settings to include the ip address as a full url and restarting octoprint. Did you do that part? Should be something like

```
http://192.168.1.2/webcam/?action=stream
```

BeerGeekGamer, 2018-08-28:
I double checked my RTMP Streamer Settings and everything looks ok. I re-ran the following test command and the stream was able to start, but was extremely unstable.

`docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.150/webcam/?action=stream 640x480 30 rtmp://live-ord.twitch.tv/app/********************************* null`

BeerGeekGamer, 2018-08-28:
I figured out what I was doing wrong. I miss read your instructions. I did not change the Stream URL in the Webcam & Timelapse, I thought you meant that, that was for the View URL section in your URL.

So far seems like the best I can do is 10fps no matter what resolution I set it to. I did see your comment under your YouTube plugin about editing the octopi.txt and entry.sh file. I'm going to mess with that and see if it does anything.

BeerGeekGamer, 2018-08-28:
I messed around with the octopi.txt and entry.sh files, but didn't seem to do anything. No mater what resolution I set, the best FPS I can do is 10.

jneilliii, 2018-08-29:
Yeah, I think it's a physical limitation of the pi + wifi (assuming this is what you are using) +  re-encoding the mjpg stream.  There probably could be improvements using the onboard gpu with -vcopy option on the ffmpeg command line with the camera as direct input vs the mjpg stream.  The only problem with that solution is that it requires direct access to the camera and so the mjpg stream service (webcamd) has to be stopped first.  I've recently seen a pull request to OctoPrint that had options for h264 streams in webcam & timelapse settings but when I attempted to test it failed to load. I'll dig into that code to see if there are possible parts that I could use to take over the internal webcam view on the control tab.

I'm glad you at least got it streaming, albeit with lower fps.

closed 2018-08-29 by jneilliii