jneilliii/OctoPrint-RTMPStreamer #8: Stream stops mid print

Celtaa, 2018-09-10:
Hi, 
 
Today I configured the RTMPStreamer plugin. 
I followed the instructions, tested all ok, configured webcam with actual ip, configured plugin to start stream with print and as soon as I started printing the stream started.
All went well for 40 minutes approximately but then the stream stopped, the print still continues and on the #tab_plugin_rtmpstreamer the button says "Stop".
Any ideais on why it stopped? Don't know if is there a log I can look at. 
Thanks!

jneilliii, 2018-09-10:
Nothing I can think of off the top of my head outside of possible network disconnects or just overall processing power.  What model Pi are you running OctoPrint on?

Celtaa, 2018-09-10:
I have a Pi 3 B+, don't know if there is a processing power there.
Regarding the internet connection, I sometimes get disconnects, does the plugin try to reconnect or something?
I'm using the pi with wifi and I was having and average of 683kbps

jneilliii, 2018-09-10:
Pi 3 B+ should be fine.  Just wasn't sure if you were using a single core Pi Zero W or something similar.  I'm not really sure if it will reconnect or not, but you could press the stop button and then press the start button to reinitialize.  Check to see if there are any errors in your octoprint.log too,

closed 2019-12-07 by stale[bot]