jneilliii/OctoPrint-RTMPStreamer #16: Cannot Start RTMPstreamer plugin

royalba94, 2019-01-08:
Hello

I'm running OctoPrint on windows 10 currently (I know, not ideal) and I can't get the plugin to load and there don't seem to be any instructions for setup on windows.

Here is what shows up in the OctoPrint terminal window:

```
2019-01-08 12:09:12,128 - octoprint.plugin.core - ERROR - There was an error loading plugin rtmpstreamer
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\plugin\core.py", line 980, in load_plugin
    plugin.load()
  File "c:\octoprint\venv\lib\site-packages\octoprint_rtmpstreamer\__init__.py", line 134, in __plugin_load__
    __plugin_implementation__ = rtmpstreamer()
  File "c:\octoprint\venv\lib\site-packages\octoprint_rtmpstreamer\__init__.py", line 16, in __init__
    self.client = docker.from_env()
  File "c:\octoprint\venv\lib\site-packages\docker\client.py", line 81, in from_env
    **kwargs_from_env(**kwargs))
  File "c:\octoprint\venv\lib\site-packages\docker\client.py", line 38, in __init__
    self.api = APIClient(*args, **kwargs)
  File "c:\octoprint\venv\lib\site-packages\docker\api\client.py", line 131, in __init__
    'Install pypiwin32 package to enable npipe:// support'
DockerException: Install pypiwin32 package to enable npipe:// support
```

___________

I'm guessing I need to install docker or something too? I already did `pip install pypiwin32` earlier (and it said it was successful) tho it doesn't seem to have changed anything.

if I can't get the streaming out to work that's fine, all I really want to do is change the webcam url (in the control section) to use a twitch/mixer/youtube stream instead of requiring an mjpeg stream (which I only just got working lol)

Hopefully I did this right, I'm new to all of this. Thanks!

jneilliii, 2019-01-08:
So the core reencoding used from the docker image stuff (ffmpeg) described in the install instructions is not compatible with windows deployment as far as I know. I believe that's pi specific as it uses a precompiled version of ffmpeg for the pi. I didn't create that part, just integrated it into a plugin. I would have to develop/program a windows solution.

jneilliii, 2020-03-02:
I think you can achieve this option better with the webcamstreamer plugin.  It has docker images for various CPU architectures and should be able to be run on windows.

https://plugins.octoprint.org/plugins/webcamstreamer/

closed 2020-03-02 by jneilliii