jneilliii/OctoPrint-RTMPStreamer #17: HTTP Err 401

Sparkyz92, 2019-01-08:
HTTP error 401 Unauthorized http://10.0.0.135/webcam/?action=stream: Server returned 401 Unauthorized

is Haproxy or something getting in the way? how can I help you help me?

Sparkyz92, 2019-01-08:
yes, followed HTTP authentication on haproxy Foosel's guide, is the line in there about webcam causing the issue? I'm using an insecure-password.

frontend public
        bind *:80
        bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem
        option forwardfor except 127.0.0.1
        acl Auth_OctoPrintUsers http_auth(OctoPrintUsers)
        http-request auth realm OctoPrint if !Auth_OctoPrintUsers
        use_backend webcam if { path_beg /webcam/ }
        default_backend octoprint
        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

jneilliii, 2019-01-09:
Can you link me the instructions you followed?  From what I can tell with that haproxy config it looks like you are applying auth to the whole realm, not just the backends.  There may be a way to bypass that for local access from the pi itself or there might be a way to embed the login information in the webcam url like `http://<username>:<passowrd>@10.0.0.135/webcam/?action=stream`, I just don't know and am not familiar enough with haproxy to give you a definitive answer, but that is definitely the problem.

Sparkyz92, 2019-01-09:
https://discourse.octoprint.org/t/how-to-enable-http-authentication-on-haproxy/180
I read down in the comments i think I found something that will work im going to try. 
I did try to put the auth info before the ip just like that before i wrote this but it didn't work. 
you're right about the backends and i think thats the solution. Thanks for coding the plugin :3

jneilliii, 2019-01-09:
Yep, the comment you mentioned was going to be my next recommendation.  Let me know how it works out for you in case someone else asks about it.

closed 2019-01-10 by Sparkyz92