jneilliii/OctoPrint-RTMPStreamer #18: Ubuntu 18.x Docker Error

schnello, 2019-02-18:
Hello.

Iam sorry but i have no experience with docker. 
Iam using Octoprint at a Ubuntu 18.x Server. Docker is installed but the "images was not created:
If i try to stream i get only the error message:
[image]

I try to build the image: At the end is one error message:
```
nas@uplink:~/rtmpstreamer$ docker build -t octoprint/rtmpstreamer .
Sending build context to Docker daemon  56.32kB
Step 1/3 : FROM alexellis2/streaming:07-05-2018
07-05-2018: Pulling from alexellis2/streaming
b906b103c0d7: Pull complete
d2cb0e116c35: Pull complete
bdc397860cef: Pull complete
8d6c3df98ed8: Pull complete
6d96283e4796: Pull complete
251e2212d0b5: Pull complete
168e665517bb: Pull complete
d941c93a358d: Pull complete
17a5c730439f: Pull complete
2dd0d3fc92c5: Pull complete
8c86036bd422: Pull complete
1ecc4deb91bb: Pull complete
49800e533d71: Pull complete
7961560d62c7: Pull complete
e602239716dd: Pull complete
12fcd33ca01e: Pull complete
e253f062764d: Pull complete
c176879df3d1: Pull complete
f831db4762ef: Pull complete
Digest: sha256:ef275dd996fd941a70efeda19267bf0da86e98f35843f62f27f1efe4d8b849fa
Status: Downloaded newer image for alexellis2/streaming:07-05-2018
 ---> a8d02138647f
Step 2/3 : COPY entry.sh entry.sh
 ---> 085858c93052
Step 3/3 : RUN chmod +x entry.sh
 ---> Running in 0d14cfa4c867
standard_init_linux.go:207: exec user process caused "exec format error"
The command '/bin/sh -c chmod +x entry.sh' returned a non-zero code: 1
```

The list of docker images:
`
nas@uplink:~/rtmpstreamer$ docker images
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
<none>                 <none>              085858c93052        6 minutes ago       1.32GB
alexellis2/streaming   07-05-2018          a8d02138647f        9 months ago        1.32GB
`

The list of container:
`
docker container ls
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
`

So the docker image is not created or? Any hint to solve this problem?

jneilliii, 2019-02-18:
I would suggest switching to the webcamstreamer plugin, many more options and easier docker instructions.

schnello, 2019-02-18:
Ok.Thank you

closed 2019-02-18 by schnello

jneilliii, 2019-02-18:
I was thinking about this a bit more and don't think the webcamstreamer plugin will help you either actually, because all of these plugins are assuming a pi device, and the underlying docker image for ffmpeg is compiled for that type of device.

schnello, 2019-02-18:
I think so too. I rename the docker image to the correct name but the system shows me now an error message that the command is not compatible. 

Docker looks nice but looks a bit like a step backwards. I take now your ffmpeg command from the entry script and put it into a screen server. Works fine.