jneilliii/OctoPrint-RTMPStreamer #20: Error while building docker

coolen555, 2019-10-11:
Followed instructions but get this error everytime i try to build the docker image.
failed to register layer: Error processing tar file(exit status 1): unexpected EOF

jneilliii, 2019-10-11:
You may want to try the webcamstreamer plugin, it's better than mine and has a docker image registered so easier to install.

coolen555, 2019-10-11:
I tried that one but i only got a black screen when streaming. So far i haven't found a single working streaming plugin so i was hoping yours would work.

jneilliii, 2019-11-30:
So you're getting the error when you run the line `docker build -t octoprint/rtmpstreamer .` Make sure you include the `.` at the end of the command line while in the rtmpstreamer folder.

closed 2019-12-21 by stale[bot]