jneilliii/OctoPrint-RTMPStreamer #23: Starting stream gives an error but not enough info about what the problem actually is

azzuron, 2020-04-29:
('Connection aborted.', error(2, 'No such file or directory'))

every time i start a print or start a stream i get this. I dont know if its an issue getting the cam feed or sending to twitch? I think all my settings are good for the sending side, and the camera works on the control tab. I also use octolapse...which works ok. I do use the http://<IP>:PORT/webcam path in my camera config for stream. Thanks!

jneilliii, 2020-04-29:
That error typically means you didn't install the docker parts?

azzuron, 2020-04-29:
yes i did realize that, but ... My octoprint is running in a docker already and putting docker into the docker manually i dont know if it would work and even if so, it would get blown away next time my octoprint docker updates. My docker for octoprint already has ffmpeg in it. can we just use that?

jneilliii, 2020-04-29:
Interesting set-up. Not with the plugin, but you could in theory use the same ffmpeg command the plugin is using in its docker to do the same thing, you would just attach that command to an [event](https://docs.octoprint.org/en/master/configuration/config_yaml.html#events), ie [PrintStarted](https://docs.octoprint.org/en/master/events/index.html#printing) and then stop it with [PrintDone](https://docs.octoprint.org/en/master/events/index.html#printing).

jneilliii, 2020-04-29:
The command specifically is below that I'm using.

```
ffmpeg -re -f mjpeg -framerate $3 -i $1 -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -s $2 -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate $3 -f flv -filter:v $5 $4
```

$1 = OctoPrint Webcam Stream from settings
$2 = Stream Resolution
$3 = Framerate
$4 = Stream URL posting to, ie twitch live stream URL

azzuron, 2020-04-29:
Hmm ok, i will have to look into that then! :) Thanks much. I didnt even know you could trigger commandline jobs from events like that. Can probably do some fun stuff with that.

jneilliii, 2020-04-29:
Yeah, there's lots of little tricks that you can do in OctoPrint just from changes to config.yaml. [Event](https://docs.octoprint.org/en/master/configuration/config_yaml.html#events) triggering, [System](https://docs.octoprint.org/en/master/configuration/config_yaml.html#system) commands (show up in the system menu), and [Custom Controls](https://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls), are the big areas where you can do this kind of stuff. There are plugins to be able to edit those last two from within the web interface, https://plugins.octoprint.org/plugins/customControl/ and https://plugins.octoprint.org/plugins/systemcommandeditor/. You could also use https://plugins.octoprint.org/plugins/gcodesystemcommands/ to create custom commands to be triggered by gcode commands, so for example you can have OCTO1 start the stream and put it into your start gcode and then OCTO2 stop the stream.

azzuron, 2020-05-02:
Cool, thank you. I checked them out but documentation on them is not detailed enough and i wasnt sure what i was doing...anyway the events in the config file are working. I ended up needing a different ffmpeg command to get it streaming smooth and clean. i removed the -re flag as based on my reading when streaming for a direct device like I am, it would cause issues and it was causing about every 2 seconds a skip in the stream. Thanks for all your help!

closed 2020-05-02 by azzuron