jneilliii/OctoPrint-RTMPStreamer #24: Twitch embed is misconfigured

0MechaGojira0, 2020-12-05:
Here is the new Twitch embed config. I dont think we have to add all of it aside from the web address. I understand how the  old embed system worked but I am lost as to what to put as  the "parent" address. 

<iframe
    src="https://player.twitch.tv/?<channel, video, or collection>&parent=streamernews.example.com"
    height="<height>"
    width="<width>"
    frameborder="<frameborder>"
    scrolling="<scrolling>"
    allowfullscreen="<allowfullscreen>">
</iframe>

jneilliii, 2021-02-03:
So I have played with this a little bit and think this will be a difficult change to work around. Based on the docs it seems that you have to include an attribute and query parameter in the src url of an iframe called `parent` which appears to require a fully qualified domain name. If that is true then there is not going to be much I can do to get around this one. I'll keep looking for possible ways around this new imposed limitation. If I can get it to work with an internal name or ip address, then I can definitely make the necessary changes to the plugin to allow those `parent` requirements to be filled in where they need to be.

0MechaGojira0, 2021-02-03:
Awesome thanks for looking into it!I appreciate it!

jneilliii, 2021-02-03:
So I just tested and I think my suspicions are correct. I can get the URL to work in conjunction with the ngrok tunnel plugin and setting the `&parent=29ce4cda33a4.ngrok.io` to the end of the player url and it does load the embed when I access the site via the ngrok tunnel url directly, ie https://29ce4cda33a4.ngrok.io in the case above. I would suspect this would also work with the octoeverywhere plugin as well. For now, that's a potential work around to the embedded player, or just have another tab opened watching through the twitch dashboard.

jjosephliu, 2021-12-08:
`https://player.twitch.tv/?autoplay=false&channel=YOURCHANNEL&parent=MYDOMAIN.com`

replace YOURCHANEL with your channel and MYDOMAIN.COM with your own FQDN

jneilliii, 2021-12-08:
Yes, that is correct but it only works with a publicly available FQDN, you can't use something like octopi.local or the ip address of the printer for that parent querystring parameter.

TheSin-, 2022-01-17:
there is one notable exception here.

if octoprint is running on your local machine (you can use 127.0.0.1 to access it) then you can use 127.0.0.1 as the FDQN and it works.  But that is the only exception I can find and it's likely fairly rare that octoprint will be running on the system you are using to access it.  But thought I'd log it.

glserr, 2022-08-18:
Hello was there a fix for this issue or is the plugin dead ?

jneilliii, 2022-08-18:
The plugin can still stream to Twitch, it's just the embedded viewer in OctoPrint can't load without the proper public fully qualified domain name using your own or octoeverywhere URLs.

jneilliii, 2024-08-10:
Since there's nothing I can really do to resolve this requirement by twitch embed closing ticket.

closed 2024-08-10 by jneilliii

TheSin-, 2024-08-10:
sadly this one is out of our control I agree, though it is sad.

jneilliii, 2025-12-06:
@TheSin- figured out a way to get this to work @0MechaGojira0 by updating the ssl certificate on OctoPrint's haproxy configuration. 

replace all octopi.local with your pi's <hostname>.local address in the command below.
```
openssl genrsa -out ~/octopi.local.key 2048
openssl req -new -x509 -sha256 -key ~/octopi.local.key -out ~/octopi.local.cer -days 365 -subj /CN=octopi.local -addext "subjectAltName = DNS.1:octopi.local"
cat ~/octopi.local.key ~/octopi.local.cer > ~/octopi.local.pem
sudo cp ~/octopi.local.pem /etc/ssl/
sudo nano /etc/haproxy/haproxy.cfg
```
look for ssl crt bind and update it to match new certificate
> redirect scheme https code 301 if !{ ssl_fc }
    bind :::443 v4v6 ssl crt /etc/ssl/octopi.local.pem

ctrl+s to save, ctrl+x to exit
```
sudo systemctl restart haproxy
```

then for the viewer url in the plugin's settings add the `&parent=octopi.local`