jneilliii/OctoPrint-RTMPStreamer #25: Does't seem to work on Pi version 4B 8GB

CobaltEcho, 2021-01-02:
I've been trying to install this on my Pi (version below), and after a while I found that OctoPrint says it's incompatable.  Any idea why, or what I need to do?

Raspberry Pi 4, Model B, Rev 1.4, 8BG RAM
OctoPi Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.4
OctoPrint Version 1.5.2

[image]

jneilliii, 2021-01-02:
It's python 3 compatibility related. Because you installed octopi 0.18 is utilizes python 3 by default. If you want to test this out for me (I don't use the plugin) you can install the rc branch here to see if it works for you or not. If it does then I can release an official stable version for python 3. Copy/Paste the URL below into Plugin Manager > Get More > ...from URL and click Install.

```
https://github.com/jneilliii/OctoPrint-RTMPStreamer/archive/0.1.5rc1.zip
```

Duckle29, 2021-01-27:
I tried installing from that url, but it still somehow gets that the compatability is >=2.7, <3, even tho it's clearly set as <4 in the source
I've tried installing with `--no-binary :all:` as well, but same story

jneilliii, 2021-01-27:
Are there any errors in the octoprint.log?

Duckle29, 2021-01-27:
Okay, forcing it compatible via config.yaml, and it is indeed not compatible

```python
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 762, in _parse_metadata
    root = ast.parse(f.read(), filename=path)
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_rtmpstreamer/__init__.py", line 26
    except Exception, e:
                    ^
SyntaxError: invalid syntax
```

Updating this syntax to python3 seems to let it start. Currently installing docker to test it out

jneilliii, 2021-01-27:
cool, be happy to accept a PR against the RC branch if you do get it working. That was a slight oversight on my part.

Duckle29, 2021-01-27:
so the docker container can stream just fine, however trying to start it from the plugin doesn't seem to spawn any docker containers, and there doesn't seem to be anything thrown up in the log.

On the first try it "started" the stream and gave me a stop button, hitting the stop button gave a `404 no such container` and a container hash. Second try it just kept spinning on start.

jneilliii, 2021-01-27:
Ok, I'll have to look into it later then.

jneilliii, 2021-02-03:
This is resolved in the above commit. I've tested on my octopi 0.18 install that is Python 3 out of the box. You should be able to switch the release channel to `Release Candidate` in OctoPrint's Software Update settings and update to version 0.1.5rc2 to verify. I am still investigating #24 which has rendered the embedded viewer tab useless, but the stream does still start as expected.

Duckle29, 2021-02-03:
Awesome. Will give it a test in a day or two :)

Duckle29, 2021-02-06:
It works for streaming, preview as you said doesn't work :)
Had to install it from github directly though, as I couldn't install the stable version and then promote it to RC.

jneilliii, 2021-02-06:
Thanks @Duckle29 official release version out now with those changes integrated. I'll get the plugin repo updated so it shows as being compatible from there too.

closed 2021-02-06 by jneilliii