jneilliii/OctoPrint-RTMPStreamer #28: 404 Client Error: Not Found

epn35, 2021-03-04:
Hello,
I have been confused for a few days with Docker. I'm trying to stream with "RTMP Streamer" or "WebcamStreamer" but it's the same problem: `<404 Client Error: Not Found ("pull access denied for adilinden / rpi-ffmpeg, repository does not exist or may require 'docker login ': denied: requested access to the resource is denied ")>`

[image]
[image]

Have you ever had this problem? Can you guide me to a solution, I don't know what to test anymore ..
Thank you.

jneilliii, 2021-03-04:
I think you posted this on the wrong repo. You're using [YouTubeLive](https://github.com/jneilliii/OctoPrint-YouTubeLive), not RTMPStreamer it seems. I'll open a new issue over there for you.

closed 2021-03-04 by jneilliii

epn35, 2021-03-06:
The plugin is closed for YoutubeLive, can I restart the subject here?

jneilliii, 2021-03-06:
Are you trying to stream to YouTube or Twitch/Other Service?

epn35, 2021-03-06:
ok i have completely reinstall octoprint and install RTMP Streamer plugin.
First problem: `RTMP Streamer Error ('Connection aborted.', FileNotFoundError(2, 'No such file or directory'))`
With correct Stream URL: http://192.168.1.10/webcam/?action=stream
And .... i try to reinstall WebcamStreamer because on option advanced i can change Docker Image and container name (picture).
[image]
[image]

i have install docker by Putty and add this:
`docker pull kolisko/rpi-ffmpeg`
I dont know if its correct or not iI didn't understand everything I was doing.
I have bad frame rate but, it's work !! :D 
i try to have a better frame rate later.. 
Thanks for the help anyway @jneilliii

jneilliii, 2021-03-06:
Yeah, that's one of the reasons the webcamstreamer plugin is better than this one. It allows you to set the docker names versus being hardcoded in my plugin's.