jneilliii/OctoPrint-RTMPStreamer #32: RTMP Streamer Error

jjosephliu, 2021-12-08:
I am terribly sleep deprived so I apologize in advance.
I am also a complete beginner to networking, docker and linux in general so I also apologize for that

### My Issue:

404 Client Error: Not Found ("No such container: 09c0e1d7e42992768535d11e6d92f486efb845bc215ba34ee91078a2c05780bb

[image]

I followed all the steps from here: ["docker_isnstructions.md"](https://github.com/jneilliii/OctoPrint-RTMPStreamer/blob/master/docker_instructions.md)

### So here's what I've got:

[image]

`https://mydomain.com/webcam/?action=stream`
For obvious reasons, I have replaced my FQDN with "MYDOMAIN.com"

I know this works; I used the test tab and its totally fine

[image]

It's a little cut off so here's my "View URL"
https://player.twitch.tv/?autoplay=false&channel=MYCHANNEL&parent=MYDOMAIN.com

Again, replaced my sensitive info here

Here's what happens:

When I press the start button, it just loads forever and my stream doesn't start

[image]

Untill I refresh the page, then it turns into a stop.

[image]

Yet, my stream still isn't up

**Then when I press stop, this pops up:**

[image]

`404 Client Error: Not Found ("No such container: 09c0e1d7e42992768535d11e6d92f486efb845bc215ba34ee91078a2c05780bb`

So I ran the test provided by  [docker_isnstructions.md](https://github.com/jneilliii/OctoPrint-RTMPStreamer/blob/master/docker_instructions.md)

I ran this in the terminal:
`docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest https://MYDOMAIN.com/webcam/?action=stream 640x480 10 rtmp://live.twitch.tv/app/live_MYSTREAMKEY null `

And interestingly enough, it returns me this:

[image]

But then when I run this: 
`docker run --privileged --name RTMPStreamer -ti octoprint/rtmpstreamer:latest http://192.168.1.111/webcam/?action=stream 640x480 10 rtmp://live.twitch.tv/app/live_MYSTREAMKEY null `
_192.168.1.111 is the private IP for my Pi_

The stream starts automatically.
But then when I press the stop stream button, the stream continues and this error pops up again:
[image]
And the button does the same loading thing when I press start again.
The stream only stops when I give the kill signal command in the terminal.

**SIDENOTE:**
I am accessing the Octoprint web controls via my FQDN
When I access the Octoprint web controls via private ip {192.168.1.111) this sis what is shown on the "RTMP Streamer" tab
[image]
`player.twitch.tv refused to connect`
Though this is a small issue compared to the error and the stream not working.

Really sorry if I did/said any really stupid stuff! I'm kind of a beginner.
Feel free to correct/guide me.

Thanks again

jneilliii, 2021-12-08:
Ah, this looks like an underlying issue with the ffmpeg command that's being used in the docker container for streaming. I didn't make that underlying container so it's a matter of re-creating that container with https support option enabled. If you were to use regular http instead of https I bet it would work, but for obvious reasons (you're port forwarding) you wouldn't want to do that. If you can find an alternative docker container for ffmpeg that supports https and takes the commands the same way you could adjust the container used by this plugin to reference it instead, and I could assist you with what needs to be changed for that.

jjosephliu, 2021-12-08:
Ok...
So what should I do here?

jneilliii, 2021-12-31:
There will be a release in the near future that should make this process a little easier and removes the docker dependency as an option. I'll report back once those changes are received and merged.

closed 2022-01-08 by github-actions[bot]