jneilliii/OctoPrint-RTMPStreamer #38: MultiCam

mlee12382, 2022-01-29:
Is it possible to get MultiCam plugin support for rtmpstreamer? It would be nice to be able to use a different camera than the primary camera for those of us that are using MultiCam.

jneilliii, 2022-01-29:
You can do this by modifying the ffmpeg command under adcanced options, replacing the `{webcam_url}` token with the url for your secondary camera.

mlee12382, 2022-01-30:
Sweet got it, I had to figure out what the proper formatting for the tokens was, it didn't want to save or even start the stream with the default settings and the primary camera. I was about to send a message asking for more help but figured I would try 1 more variation and it worked lol. Thanks for pointing me in the right direction! Cheers.

mlee12382, 2022-01-30:
New issue, maybe I'm just missing something but when I reboot my pi the ffmpeg commands revert to default an no longer have my url saved?

jneilliii, 2022-01-30:
separate bug report created.

closed 2022-01-30 by jneilliii