jneilliii/OctoPrint-RTMPStreamer #39: when I reboot my pi the ffmpeg commands revert to default

jneilliii, 2022-01-30:
New issue, maybe I'm just missing something but when I reboot my pi the ffmpeg commands revert to default an no longer have my url saved?

_Originally posted by @mlee12382 in https://github.com/jneilliii/OctoPrint-RTMPStreamer/issues/38#issuecomment-1025054905_

mlee12382, 2022-02-02:
Where do the files install on the pi? I can just manually edit it in nano via ssh as a temporary fix?

jneilliii, 2022-02-02:
I'm not sure setting it manually will make a difference. I need to make adjustments to the code. I'll take a look tonight and see if there is a quick fix.

jneilliii, 2022-02-03:
Fixed with the above commit and will be released shortly as version 1.0.2.

closed 2022-02-03 by jneilliii

mlee12382, 2022-02-05:
Awesome thanks for the assistance  :)