jneilliii/OctoPrint-RTMPStreamer #44: No video output

Eruditionis, 2022-03-14:
I wasn't getting the plugin to work through octoprint, after emailing Jneilliii I figured out docker was my issue. Now, when I enable the stream through docker, I get enough to start the stream via YouTubeLive, but it's only a snapshot. I get no video.

I am using FFmpeg for streaming, settings/fails in pics
[image]
[image]
[image]
[image]
[image]

When I used docker I could get test video, but no output from the plugin, now that docker is gone, I have output from the plugin, but no video :/

jneilliii, 2022-03-15:
enable the option for ffmpeg debug and attempt to start the stream. this will give more detailed error messages from the ffmpeg command it's own logging file in the logging section of OctoPrint.

Eruditionis, 2022-03-15:
Done, I got video but it is giving me an error saying there isn't enough of
it being sent to maintain a smooth stream. It is buffering for a minute
then has sped up video, also appears to be stretched out as seen if you
click this link <https://youtu.be/vQ8fdBm6qXY>, hopefully it's still going
when you get this. I will also link the rtmpstreamer log below for you in
case it helps?

TheSin-, 2022-03-15:
The stretch is cause you aren't setting the size to the aspect of your camera, make sure you know the natural aspect of your cam when you set the stream size.

As for buffering remember you're limited by hardware or network if you're on a Pi via Wifi, that's some heavy restrictions and YouTube requires a very heavy stream to be happy.

Thought honestly your stream looks really good.

Eruditionis, 2022-03-15:
Maybe it's me being very picky, I appreciate it! How do I tell the size of
my camera? Some streams come out good, some are stretched. Maybe just
restart it until it works?

Eruditionis, 2022-03-15:
Also, I am using a Ethernet cable, before I got flawless videos(just a little grainy), but I have been streaming for 40 minutes and only have about 15 minutes worth of video

TheSin-, 2022-03-15:
I'm doing more work on the setup to enable HW encoders which will help a lot.  Right now it's using SW encoders.

As for video size are you on a Pi?

Eruditionis, 2022-03-15:
Yes, I have a rpi 4B with this camera
<https://www.amazon.com/Raspberry-Camera-Webcam-OV5647-Adjustment/dp/B08QFM8TVV/ref=sr_1_5?crid=Z3040ITF7LEX&keywords=camera+raspberry+pi&qid=1647354231&sprefix=camera+raspberry+pi%2Caps%2C64&sr=8-5>

Eruditionis, 2022-03-15:
So should I ssh into the pi to change that value? How much should I give for reference? I'm very new to this

TheSin-, 2022-03-15:
There are soooo many factors to answer that, I'd start with 128 and move up from there.

Eruditionis, 2022-03-15:
I have 128 automatically assigned, so I will bump it up to 256

Eruditionis, 2022-03-15:
> I think 256 is what I typically put in that setting.

Okay, I just changed the setting, I will sudo reboot my pi when it is done
with it's current print job. Will update as soon as possible!

Eruditionis, 2022-03-16:
256 yields the same results, should I bump it up again? For 2 min of stream time, only <15 seconds of footage was sent out

Eruditionis, 2022-03-16:
[image]

TheSin-, 2022-03-16:
turn off the overlay, it will slow things down as it needs to process each frame, see if that helps.

Eruditionis, 2022-03-16:
The overlay is off, restarted live stream, no overlay in the video, but the output is still just as bad. If I hadn't had really good streams just a week ago, I would guess my wifi is bad, but I had really good streams on this wifi not long ago

TheSin-, 2022-03-16:
Remember the plug just gives you the tools, the setup might need to be googled a bit, I'm no pro with YouTube I found it to be very difficult and lacking so I use Twitch.

Eruditionis, 2022-03-16:
i'm not sure where to find the bitrate to change that, is it in one of the config files? Or can I find it in octoprint?

Could very well be Youtube, I am going to try it on Facebook, and if no beans there then I will use twitch.

Eruditionis, 2022-03-16:
My guess is that the bitrate is 700, because I have no idea where it is so no changes to it

Eruditionis, 2022-03-16:
That's fine, I am also at work lol, that's the only time I have access to the printer (I work at a library)

I will see if I can change that, should I change anything else in that area? I notice there are a bunch of options labeled for experts only

TheSin-, 2022-03-16:
It's there for those that know ffmpeg to have more granular control.  So feel free but I'd read first ;)

The big thing with YouTube is getting the amount of data it wants, that being said the old version of the plugin was using 700k as well, none of that changed, the big change with the overlay, which is why I suggested we remove it.  Since it's acting the same I'm very curious what the issue is, I just don't know enough about YouTube but I know it's been a pain.

I plan to work more on the plugin very soon I'll try and focus some attention on YouTube maybe I can change some default settings if it's set to YouTube if Ic an find what the issue with it is.

Eruditionis, 2022-03-16:
Okay, so I just changed the setting to Youtube's recommended bitrate of 4500K and still have buffering. Funny enough it is actually much worse now...

Yeah I can try other ways to stream, our thing is to provide our patrons with the ability to watch their prints as they progress, so it doesn't matter where it streams to as long as it streams lol.

I really appreciate all this help though, I do plan on contributing to the plugin, this is one of my most used plugins

TheSin-, 2022-03-16:
https://support.google.com/youtube/answer/2853702?hl=en

TheSin-, 2022-03-16:
looks like they want keyframes which I'm not setting in the plugin give me a few minutes to hash out the string of it and I'll send you something to try.

Eruditionis, 2022-03-16:
Okay that works! Thank you so much for all of this

TheSin-, 2022-03-16:
in the experimental

can you try to replace this part `-vb {bitrate}` with `-b:v {bitrate} -maxrate {bitrate} -minrate {bitrate} -bufsize {bitrate}`

Remember if you make changes there when a new version is released it won't change that, you'll need to remember to click the resend button on that filed to reset it.  And this is just a test, I can't test it currently :)

TheSin-, 2022-03-16:
no saving and restarting the stream is good enough

Eruditionis, 2022-03-16:
So I am getting the same issue, buffering with 1-3 seconds of video in between buffers. 

I had this set up to run through docker before and had the video running, only issue was that I couldn't start the stream through the plugin. Do you think that route would be easier or should we stick with this?

TheSin-, 2022-03-16:
docker would add a layer, the only difference would be the string that starts ffmpeg if you can extract what you had for your string I could tell you how to setup the plugin to work the same.

Eruditionis, 2022-03-16:
I'm not sure if this detail matters, but when I do get video, it's very sped up.

TheSin-, 2022-03-16:
it's just catching up, it's using UDP so there is no frame order.

Eruditionis, 2022-03-16:
Just checked, I didn't reinstall docker since I reimaged the pi, will install now and update in a few mins

TheSin-, 2022-03-16:
one sec I'll check the docs

Eruditionis, 2022-03-16:
Server: Docker Engine - Community
 Engine:
  Version:          20.10.13
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.16.15
  Git commit:       906f57f
  Built:            Thu Mar 10 14:05:46 2022
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.5.10
  GitCommit:        2a1d4dbdb2a1030dc5b01e96fb110a9d9f150ecc
 runc:
  Version:          1.0.3
  GitCommit:        v1.0.3-0-gf46b6ba
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

Eruditionis, 2022-03-16:
I finished installing docker, I also had it pull kolisko/rpi-ffmpeg:latest

Eruditionis, 2022-03-16:
How do I reset the ffmpeg command? 

I replaced that.

When I put that command in, I get this: (I didn't leave the xxxx-xxxx's in there)

` docker run -it --rm --privileged adilinden/ffmpeg ffmpeg \ -re -f mjpeg -framerate 5 -i http://192.168.1.16:8080/?action=stream \ -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k \ -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 \ -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx-xxxx
Unable to find image 'adilinden/ffmpeg:latest' locally
docker: Error response from daemon: pull access denied for adilinden/ffmpeg, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.`

TheSin-, 2022-03-16:
you can't replace it with that.

there is a button right beside the title.  Also if you ran that you'll need to reboot now.

Eruditionis, 2022-03-16:
I just rebooted the pi, I thought it failed when I tried running that command? 

I meant that I replaced the  {videocodec} with h264

TheSin-, 2022-03-16:
that's cause you are running it with docker, don't use docker.

Eruditionis, 2022-03-16:
Okay, I’ll shut off docker my bad, will run again when I’m back at work
tomorrow and let you know ツ

jneilliii, 2022-03-17:
not sure how tight your library's network is, but you might want to look into something like tailscale to be able to remote access to your pi/OctoPrint remotely. You basically install on the pi and then install on whatever device you want to be able to access it from and it will establish a connection between the two. I've used it in the past and it was pretty easy to setup. Another option if you only need to access OP is OctoEverywhere plugin.

Eruditionis, 2022-03-17:
I just got the octoeverywhere plugin, all set up to go so I can change op settings from wherever. Can I use that tailscale program to perform ssh functions from home too?

Eruditionis, 2022-03-17:
I don't have docker checked, I can't get any video to go through now :( 

Should I remove docker from the pi?

Eruditionis, 2022-03-17:
Weirdly enough, I can't do those commands, they all ask for a password, when I do type it in they tell me I cannot use that type of verification anymore,

Furthermore, when I use the [link provided](https://github.com/adilinden-oss/docker-ffmpeg.git), they do not exist anymore

jneilliii, 2022-03-17:
Yes, tailscale is like a VPN solution that will allow you to do anything with the pi, including ssh. You just have to utilize the pi's tailscale ip instead of the local network one.

Eruditionis, 2022-03-17:
Oh okay cool, i'm definitely gonna look into that!

Eruditionis, 2022-03-23:
I ended up getting a streaming software and inputting everything manually. Thus I do not require the plugin anymore, as an added bonus I get sound to the background of my stream and better quality. Thank you very much for your help!

closed 2022-03-23 by Eruditionis