jneilliii/OctoPrint-RTMPStreamer #45: Not seeing YouTube live stream

andyylin, 2022-06-20:
I've got everything set up, and at one point it was working before, but now even though the logs report successful streaming, no video is showing up on YouTube live.

I've checked that the stream keys are correct, and formatted based on the URLs provided as hints in the settings pages.

```
2022-06-20 08:47:41,368 OctoPrint-RTMPStreamer loaded! Checking stream status.
2022-06-20 08:47:41,389 stream is inactive
2022-06-20 10:03:32,971 Launching ffmpeg locally:
|  /usr/bin/ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.1.116/webcam/?ac$
2022-06-20 10:03:33,086 Stream ffmpeg pid 3692
2022-06-20 10:03:33,086 Stream started successfully
2022-06-20 10:03:33,104 Starting dynamic overlay loop
2022-06-20 10:36:03,061 Stream stopping ffmpeg pid 3692
2022-06-20 10:36:03,063 Stopping dynamic overlay loop
2022-06-20 10:36:03,066 Stream stopped successfully
```

jneilliii, 2022-06-20:
> YouTube is not receiving enough video to maintain smooth streaming. As such, viewers will experience buffering.

this indicates that it is sending, and I've always gotten that message even with my other YouTube live plugin. To debug this will probably need to enable the debug logging for ffmpeg in the settings to get detailed messages from ffmpeg output. It will generate a separate log file.

andyylin, 2022-06-22:
Here's a log from this morning so far:

```
2022-06-22 07:40:34,289 Launching ffmpeg locally:
|  /usr/bin/ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.1.116/webcam/?action=stream  -f lavfi -i anullsrc -acodec aac -ab 128k -s 1280x720 -vcodec libx264 -threads 1 -pix_fmt yuv420p -framerate 5 -g 10 -vb 400k -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/51kr-urxb-rzyb-zt7f-0wy6
2022-06-22 07:40:34,381 Stream ffmpeg pid 5713
2022-06-22 07:40:34,382 Stream started successfully
2022-06-22 07:40:34,386 Starting dynamic overlay loop
ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable [... cut ...]
frame=   11 fps=5.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
frame=   12 fps=4.6 q=22.0 size=      55kB time=00:00:00.00 bitrate=N/A speed=   0x    
frame=   16 fps=4.9 q=29.0 size=      61kB time=00:00:00.40 bitrate=1253.3kbits/s speed=0.124x    
frame=   18 fps=4.7 q=27.0 size=      73kB time=00:00:00.80 bitrate= 750.8kbits/s speed=0.207x    
frame=   21 fps=4.5 q=23.0 size=      82kB time=00:00:01.40 bitrate= 477.7kbits/s speed=0.303x    
frame=   23 fps=4.5 q=27.0 size=     138kB time=00:00:01.80 bitrate= 627.3kbits/s speed=0.3
...
```

jneilliii, 2022-06-22:
Could you try lowering the resolution in the plugin's settings and see if it does the same. It looks like the input is detecting 640x480 from the mjpg stream, so I'm guessing that you haven't modifed octopi.txt for higher resolution.

andyylin, 2022-06-23:
> Could you try lowering the resolution in the plugin's settings and see if it does the same. It looks like the input is detecting 640x480 from the mjpg stream, so I'm guessing that you haven't modifed octopi.txt for higher resolution.

Plugin is set to 1280x720.

Good catch, I restored a backup of OctoPrint on a new Raspberry Pi 4B but forgot to install some of the other things and settings. I've fixed the resolution to be the full Pi resolution. The stream still fails to show up:

```
2022-06-23 18:40:26,146 Checking stream status.
2022-06-23 18:40:26,148 stream is inactive 
2022-06-23 18:41:57,027 Launching ffmpeg locally:
|  /usr/bin/ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.1.116/webcam/?action=stream  -f lavfi -i anullsrc -acodec aac -ab 128k -s 640x480 -vcodec libx264 -threads 1 -pix_fmt yuv420p -framerate 5 -g 10 -vb 400k -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/XXXX
2022-06-23 18:41:57,107 Stream ffmpeg pid 2780
2022-06-23 18:41:57,108 Stream started successfully
2022-06-23 18:41:57,112 Starting dynamic overlay loop
ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
  avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls  [... cut ...]
      cpb: bitrate max/min/avg: 0/0/400000 buffer size: 0 vbv_delay: -1
    Stream #0:1: Audio: aac (LC) ([10][0][0][0] / 0x000A), 44100 Hz, stereo, fltp, 128 kb/s
    Metadata:
      encoder         : Lavc58.35.100 aac
frame=    3 fps=0.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
frame=    6 fps=5.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   0x    
[mjpeg @ 0x1167400] Thread message queue blocking; consider raising the thread_queue_size option (current value: 8)
```

jneilliii, 2022-06-23:
>     Stream #0:0: Video: h264 (libx264) ([7][0][0][0] / 0x0007), yuv420p(progressive), 640x480, q=-1--1, 400 kb/s, 5 fps, 1k tbn, 5 tbc

@andyylin it seems to still be lower res, did you restart the pi or at minimum the webcamd service after making the resolution changes in octopi.txt? 

@TheSin- any ideas on what could be happening here?

andyylin, 2022-06-24:
I did restart `webcamd` during a print, and I rebooted the whole Raspberry Pi system, and it still didn't work.

I did manage to get it to work right now for whatever reason by resetting the YouTube key and also resetting the plugin. It can stream at 640x480, will try different resolutions too.

closed 2022-06-24 by andyylin

andyylin, 2022-07-11:
Stopped working after a day or so. I've switched back to Twitch (main reason for using YouTube was I could set the stream privacy), and it's been working flawlessly on Twitch.

TheSin-, 2022-07-11:
I still plan to detonate some time to YouTube, just haven’t had time, I hope within the next month as I should be don’t my car build and it’ll be in for paint so I’ll have time