jneilliii/OctoPrint-RTMPStreamer #49: Feature Request

mlee12382, 2023-08-22:
I was looking for something similar to this that would work with Klipper without Octoprint. I'm thinking a shell script that you can call in the print_start/  print_end macros to start / stop the stream and a rtmp.cfg file that it accesses that contains the user's stream url / key and other settings like the camera device path.

Is this something you would willing to put together?

jneilliii, 2023-08-22:
I don't use klipper so not something I'll be looking into. Would recommend you look at how the plugin is doing the work. It's literally just an ffmpeg command line with a mjpg input stream (crows nest I think is what you guys use) and the relevant options for outputting to whatever service you want to use.

closed 2023-08-22 by jneilliii

mlee12382, 2023-08-22:
Thanks for the response, I will look into figuring it out, my python skills are very minimal atm, so figured it didn't hurt to ask haha. 
Are the dynamic overlays part of the ffmpeg command?

jneilliii, 2023-08-22:
Yes, the overlay is created as an image in this function.

https://github.com/jneilliii/OctoPrint-RTMPStreamer/blob/master/octoprint_rtmpstreamer/__init__.py#L405-L515