jneilliii/OctoPrint-RTMPStreamer #50: Request: Addition Encoders / Use Hardware Encoding (auto)

EndlessEden, 2023-09-18:
Issue: By default, no options are provided to instruct ffmpeg to use hardware encoding/decoding.
Solution: add drop-down box and checkbox to enable hardware encoding/decoding to ffmpeg line.

Example: 

> [x] Enable Hardware Acceleration

> (Dropdown box) {Decode}
> {Encode}

> hardware_decode/hardware_encode | -hwaccel_device=auto

Further, the available encoders should include h264_omx, h264_cuvid, h264_amf, h264_vaapi if Hhardware Acceleration is ticked and "Encoding" is selected from the dropdown box.

> 
(Example ffmpeg line with hardware accel enabled: {ffmpeg} -re -f mjpeg -framerate {frame_rate} {hardware_decode} -i {webcam_url} {filter} {hardware_encode} -f lavfi -i anullsrc -acodec aac -ab 128k -s {stream_resolution} -vcodec {videocodec} -threads {threads} -pix_fmt yuv420p -framerate {frame_rate} -g {gop_size} -vb {bitrate} -strict experimental -f flv {stream_url})

---

This type of solution makes using hardware accel easy, without needing to be a expert in the use of FFMPEG.

A simple description describing the encoders (omx = rpi, cuvid = nvidia, amf = amd, vaapi = intel/quicksync) is all thats needed to get users going. -- any encoding issues due to hardware encoding not being availible on the host os would be the responsibility of the user to resolve.

FFMPEG on Octopi-OS is already built with mmal/h264_omx built in, so not including this when native support exists is a major lost opportunity.

jneilliii, 2023-09-18:
You seem knowledgeable about this, happy to accept pull request with these changes, not sure when I'll get to it.

jneilliii, 2023-09-18:
I'd be completely happy removing the docker options entirely.

TheSin-, 2023-09-18:
That would simplify it, but then you'd be locking it to RPi only and we'd have to detect that.  As technically it can be run on other platforms currently.

jneilliii, 2023-09-18:
> That would simplify it, but then you'd be locking it to RPi only and we'd have to detect that.

You mean the acceleration options? I tested the ffmpeg option on my windows machine during development and had no issues with the non-docker implementation.

TheSin-, 2023-09-18:
yeah I mean if we add HW accel to the lists or options.  That or just like docker we only enable them on RPI but have to detect when we aren't on RPI and auto remove them.

jneilliii, 2023-09-18:
Yeah, that makes sense and the reason why I thought you made the command-line editable. I think the idea of detecting RPi and automatically adding the options to the command line is a neat idea, but if that's done why would you make it an option and not just automatically do it?

TheSin-, 2023-09-18:
also a valid point, for me if you didn't want that option for whatever reason (bad firmware causing issues) you would have no way to disable it.

I'm not against adding HW Accel as I use it myself.  But I just couldn't find a safe way to do it for all situations and not over complicating the startup by adding detection.

I did add a little magic to change the lists, and it could be used to add the options I think.

jneilliii, 2023-09-18:
yeah, I was just looking at the pi support plugin and it does have an `is_octopi` function.

TheSin-, 2023-09-18:
which I'm pretty sure is what I use to add the lists of modules.  I also detect windows and MacOS IIRC.