jneilliii/OctoPrint-RTMPStreamer #51: Bug: Unable to start stream after ffmpeg crashes during startup

EndlessEden, 2023-09-18:
Issue: FFMPEG crashed during startup due to too high of a bitrate set (was testing). However the "Start"/"Stop" button located on the RTMP Streamer tab hangs.

Suggested Solution: Have python track the PID and if it hangs, crashes or takes longer then 30s to start transmitting. consider the thread dead/hung(kill automatically) and present the start button again.

jneilliii, 2023-09-18:
happy to accept pull request with these changes, not sure when I'll get to it.

TheSin-, 2023-09-18:
This is a very tough one, the issue is that python won't really know if the stream is running, there is no way for it to know since the stream can no be monitored.  The rest is possible though.  I ran into this issue as well originally, but I thought I had solve it.  I know it's not monitored if it's in docker though.