jneilliii/OctoPrint-RTMPStreamer #55: A way to rotate video before streaming ?

tavdog, 2024-04-21:
I would like to rotate the video as my camera is at 90 degrees. I can't figure out a way to rotate it. Maybe with the ffmpeg command ?

jneilliii, 2024-04-21:
Yeah, you should be able to achieve that with vfilters (-vf) command I think. Quick search seems to indicate you need `-vf "transpose=1"`

For the transpose parameter you can pass:

0 = 90° counterclockwise and vertical flip (default)
1 = 90° clockwise
2 = 90° counterclockwise
3 = 90° clockwise and vertical flip

tavdog, 2024-04-21:
Yes! This worked.  Here is my updated ffmpeg command
`{ffmpeg} -re -f mjpeg -framerate {frame_rate} -i {webcam_url} {filter} -f lavfi -i anullsrc -acodec aac -ab 128k -s {stream_resolution} -vcodec {videocodec} -threads {threads} -pix_fmt yuv420p -framerate {frame_rate} -vf "transpose=1" -g {gop_size} -vb {bitrate} -strict experimental -f flv {stream_url}`

tavdog, 2024-04-21:
It does work but now the image is squashed.

closed 2024-04-21 by tavdog

jneilliii, 2024-04-21:
@TheSin- you maybe have any ideas about the best way to rotate?

TheSin-, 2024-04-21:
I would have thought the same though I would have though 3 not 1.

That being said does the input and output match in res? it's odd that it's squashing it.

tavdog, 2024-04-22:
Setting the resolution to be 480x640 unsquashes it.

TheSin-, 2024-04-22:
so it's working then?

tavdog, 2024-04-22:
Yes it's working now.

TheSin-, 2024-04-22:
excellent, happy to hear it.